ViennaMike/ChatterPi #1: Could I use and i2c servo controller?

czadikem, 2020-12-15:
Could I use something like an Waveshare Servo Driver HAT 16-Channel 12-bit I2C Interface or an Adafruit 16-Channel PWM / Servo HAT with this code?

ViennaMike, 2020-12-26:
Yes, you should be able to do that. Do you want to drive the jaw through such a device, or just the other skull or prop motions? To drive the jaw through it, you'll have to go through any of the code in the AUDIO class in audio.py dealing with the jaw and rework it from a direct pwm out to a servo (where I use the GPIO Angular Servo library) to instead send the appropriate setting through the I2C interface to the hat.

To control ONLY other servos, you'd need to add new modules for controlling their motions, as the current program only commands the jaw servo. But that should certainly be possible, and would be a nice all in one implementation, doing away with the Maestro servo controller. I wouldn't try to do it on a Pi Zero, as it's already being driven close to its max to handle the audio processing, but any other current or recent model Pi should work. 

To control BOTH the jaw and all the others, you'd need to do both of the above.

closed 2021-01-13 by ViennaMike

srvoets, 2024-10-13:
> Yes, you should be able to do that. Do you want to drive the jaw through such a device, or just the other skull or prop motions? To drive the jaw through it, you'll have to go through any of the [...]

This answered my question from issue #20