ViennaMike/ChatterPi #2: Issues when running main.py

Montanadude, 2020-12-16:
I have tried your setup on a RPI 3a+ using Buster and have all package requirements installed.  Alsamixer shows my default correctly and I can use aplay to play a .wav just fine.  

When I run python3 main.py I'm getting a 
'AUDIO' object has no attribute 'play_audio'  and nothing plays.  

Any ideas.

ViennaMike, 2020-12-18:
Nothing pops immediately to mind. I plan to look some more at it and get
back to you, unfortunately I've had a (hopefully minor) medical issue pop
up, so I can't do it right away.

czadikem, 2020-12-18:
Are you using, stretch, buster, jessie,or wheezy?  We have tried buster and stretch and have the same issue's.

ViennaMike, 2020-12-26:
I'm running buster with no problem: 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

ViennaMike, 2021-01-13:
The bug fix has been tested and posted, and I have heard no more about issues, so I'm closing this issue.

closed 2021-01-13 by ViennaMike