ViennaMike/ChatterPi #3: Unable to get Chatterpi to Use Microphone - Stops execution

naota2k3, 2021-05-18:
I am working on a Proof Of Concept for a cosplay prop using chatterpi. My goal is to test using a Raspberry pi 3B with a Tower Pro MG90s Micro Servo to move a Jaw on a Wearable Skull when I talk. I have it wired to the proper GPIO Pins. I am using a UGreen USB sound card. The Sound card is connected to a Throat Microphone so that no external sounds can be picked up outside of my voice. There will be no audio output as I want to have the prop move when I talk. 
The Pi detects the USB Sound Card (I have tried multiple sound cards.) I did disable the built in sound card  by adding blacklist snd_bcm2835 to the modprobe.d directory under the alsa-blacklist.conf file. The alsa-base.conf file shows options snd-usb-audio index=1
 
```
pi@raspberrypi:/etc $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=lan78xx, 480M
            |__ Port 2: Dev 6, If 0, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 2: Dev 6, If 1, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 2: Dev 6, If 2, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 2: Dev 6, If 3, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
```

Alsa Mixer shows the PulseAudio card and I can set it to the USB sound card.
I can use APLAY to record and playback sound files using the command prompt. 
Audacity can utilize the sound card as well. 

I have followed the instructions in the README.md file. 

```
$sudo apt-get update
$ sudo apt-get upgrade
$ sudo apt update
**Then install the package for Python 3:**
$ sudo apt install python3-gpiozero
**To install PiGPIO, type:**
$ sudo apt install pigpio
$ sudo systemctl enable pigpiod
$ sudo apt-get install python3-pigpio _The first command downloads and install PiGPIO, while the second command ensures that it starts running in the background whenever the Pi starts. The 3rd command loads the required python package._
**To install PortAudio**:
$ sudo apt-get install portaudio19-dev
**To install PyAudio**
$ sudo apt-get install python3-pyaudio
**To install Scipy, use the command:**
$ sudo apt-get install python3-scipy
```

I have also installed the following packages 
```
$ sudo apt-get install libportaudio0 libportaudio2 libportaudiocpp0
And $ sudo pip3 install pyaudio to ensure audio libraries are installed.
```

I have copied the contents of chatterpi_v0.9 into the /home/pi/chatterpi directory and I have copied the ambient and vocals directory to the chatterpi directory.

```
pi@raspberrypi:~/chatterpi $ ls -la
total 92
drwxr-xr-x  5 pi pi 4096 May 18 16:09 .
drwxr-xr-x 20 pi pi 4096 May 18 16:04 ..
drwxr-xr-x  2 pi pi 4096 May 16 22:33 ambient
-rw-r--r--  1 pi pi 8062 Dec 26 23:35 audio.py
-rw-r--r--  1 pi pi 7037 May 18 14:48 audio.pyc
-rw-r--r--  1 pi pi  537 Dec 26 23:35 bandpassFilter.py
-rw-r--r--  1 pi pi 1061 May 18 14:49 bandpassFilter.pyc
-rw-r--r--  1 pi pi  462 May 18 16:06 config.ini
-rw-r--r--  1 pi pi  451 May 16 22:36 config.ini.back
-rw-r--r--  1 pi pi 1819 Dec 26 23:35 config.py
-rw-r--r--  1 pi pi 1504 May 18 14:48 config.pyc
-rw-r--r--  1 pi pi 6978 Dec 26 23:35 controlPanel.py
-rw-r--r--  1 pi pi 2877 Dec 26 23:35 control.py
-rw-r--r--  1 pi pi 2451 May 18 14:48 control.pyc
-rw-r--r--  1 pi pi  337 Dec 26 23:35 main.py
-rw-r--r--  1 pi pi 1685 Dec 26 23:35 maxVol.py
-rw-r--r--  1 pi pi    0 May 18 16:09 output.txt
[... 213 lines cut ...]
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/home/pi/chatterpi/audio.py", line 117, in cleanup
    normalEnd()
  File "/home/pi/chatterpi/audio.py", line 114, in normalEnd
    self.jaw.angle = None
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 432, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 148, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1788, in angle
    self.value = None
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 432, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 148, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1627, in value
    self.pwm_device.pin.frequency = None
AttributeError: 'NoneType' object has no attribute 'frequency'
pi@raspberrypi:~/chatterpi $

```
when I configure the config.ini to use source = files, ambient = on, prop_trigger = timer,  the ambient file plays,  then the audio will start to play and the servo will move. The audio sounds very garbled, and cuts in and out.  

I don’t know what is going on and would like to see if I may have configured the pi wrong or if I  am doing something incorrectly. Any help would greatly be appreciated.

ViennaMike, 2021-05-19:
Thanks for your extensive documentation on the problems you're seeing! I'll try to take a look at it over the weekend.

ViennaMike, 2021-06-03:
I just wanted to give you an update. Strangely, I can reproduce a similar issue (although it mostly plays my audio message, with garbling only at the beginning) when I install and run on a Pi 4 (but it worked on a Pi Zero W, which should have had the same issue only worse. You get way more ALSA errors than I get, I'm not sure why, plus you get Jack errors that I don't get (I don't know why yours has Jack trying to run at all). Did you start from a clean Pi install, or were there other things and audio settings already on the Pi? Because I also get 
`ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
`
when I run it and get garbled output, but I get a bunch of them and the Pi recovers and keeps running. I can't reproduce the program crashing at that point. 

In a perfect world, we woudn't be getting the underrun errors, I'll try a few more things to see if I can elliminate them.

naota2k3, 2021-06-04:
I have the Raspberry Pi B+, not overclocked running a fresh install of Rasbian on a 128 gig Microsd Card. I have a copy of Rasbian on a Noobs SD but I was going to reserve that for a PI Zero so I can install it into the skull in which I am going to be wearing and have it move when I talk.   This is the 3d printed Skull Mask I am going to be using. https://www.youtube.com/watch?v=weCU1Pv9C8o  

If needed I can solder on the header and try the RaspberryPi Zero with the Noobs Installed SD CARD. and try seeing if i get different results with the Chatter Pi.   I am not too worried about the audio playing as I won't have any speakers but I am trying to have My Microphone which is a throat microphone  be used as the input and as I speak it would move the servo in sync to the voice.

ViennaMike, 2021-06-06:
Still working on it. It's really strange. There IS one bug for using sound files that I found, but that's not the main problem. In control.py, add () to the end of line 37 (should be a.play_vocal_track() ). 

But the core problem seems to be that somehow the callback function that analyzes the sound chunk and drives the jaw completes in sufficient time on a Pi Zero (and I think on my Pi 3, haven't rechecked), but not on my Pi 4 or your Pi B+. This makes little sense to me. However it may be related to the CPU governor. If I install rpi-cpu.gov (https://github.com/DavidM42/rpi-cpu.gov) and set the governor to "performance" my Pi 4, it doesn't work perfectly, but works better for a sound file, with less overruns and only a little garbling at the start, but it still crashes using MICROPHONE mode (which is what you care about). Note: changing the governor does NOT overclock the Pi, just keeps it from using a power-saving mode, so it shouldn't hurt anything. 

You can try those two things to see if they help on your Pi B+. I'm also going to try to refactor my code to use blocking rather than callbacks. I don't know if this will work at all for this functionality, but I'll check it out this week and see if that still works and eliminates the problem.

andreasba1206, 2021-08-07:
Hello Mike,

First I would like to Thank you for the project. Thats exactly what I was looking for.

So after finding, I wanted to test this program on my Pi.
As first I have tried this on a Raspberry 3B, but I had issues with the microphone mode as naota2k3.
Then I ordered an Raspi zero W, because I have seen that this was working fine on your side.

So I tried to set it up according your description. 
Everything worked fine (much better as on the Rasp 3B).
Only 1 additional reboot during the pigpoid installation was needed ;) 
I also added the () in the control.py file.

But finally it does not work (The PIR seems to work).
I will get following error

```
pi@ChatterPi:~/chatterpi $ python3 main.py
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... 23 lines cut ...]
[AUDIO]
buffer_size = 16384
source = MICROPHONE
output_channels = BOTH
mic_time = 30
ambient = ON

[PROP]
prop_trigger = START
delay = 10
eyes = ON
trigger_out = ON

[PINS]
jaw_pin = 18
pir_pin = 23
eyes_pin = 25
trigger_out_pin = 16

```

Please could you help me?

Thanks in advance
Andy

andreasba1206, 2021-08-07:
Hello Mike,

I have found out the issue! But I am not sure if it's the same for naota2k3.

I had already added the () in line 31 of the control.py
After checking your code, I found also missing () in line 51 of the control.py

Thanks again!

Andy

ViennaMike, 2021-10-04:
The missing () have been fixed in the latest update (10_3_2021).

closed 2021-10-04 by ViennaMike