ViennaMike/ChatterPi #4: ALSA buffer underruns on some models of Raspberry Pi

ViennaMike, 2021-06-11:
This is a strange one. Version 0.9 runs fine on a Raspberry Pi Zero W, which one would think was relatively slow, but generates many ALSA buffer underrun errors on a Pi4 and reported on a Pi3B. 

The issue is clearly related to taking too long in the audiopy callback function that determines the jaw movements, but I have not found a way to make that run any faster. 

Increasing the buffer size to 16384 seems to elliminate the problem on the Pi4. 

Need to update the documentation to reflect this.

GuyFleeqman, 2021-09-13:
Hi Mike,
I also had this problem.  Running a new Pi 3 model A+, with adafruit usb audio adapter.  Program runs, but got MANY ALSA buffer underrun errors.
Additionally and interestingly, the audio from the example wave file as well as additional .wav file are distorted.  The v01 poem file, the first 2/3 of the audio is 'clicky'...that is, you can tell that the clicks are following the spoken word, but its not intelligible.  Then, it switches for the last 3rd, and the speech is fine.  A second wav file, which just has a ringing bell sound, is also clicky, and does not sound correct at all.  I'm going to implement the buffer size change, but appreciate any thoughts otherwise.

Steve

ViennaMike, 2021-09-16:
I'm going to start getting back to looking at this tomorrow. I have some
ideas but need to test them out. The way Rasbian handles audio changed a
few versions ago, and I suspect that might be causing the timing problems
that cause the buffer underrun errors. If so, it may be fixable by using an
old version of the OS, but I haven't yet had a chance to test this
hypothesis.

GuyFleeqman, 2021-09-16:
Hi Mike!
Hey, I'm new to github, but thanks for the response!  I'm not an expert by
any means at Python code, so I'm not going to tinker with the
code...although I'm trying to go through it to try to understand how all
the pieces fit together.  Meantime I did order a zero to see if works on
that board.

Best regards,

Steve

jamesluke32, 2021-10-03:
Thanks for working on this it’s awesome. I’m in the process of making a talking Audrey 2 using chatterpi. Have you been able to make any progress on the playback issue? I’m running a zero w and am also experiencing the problem.

ViennaMike, 2021-10-03:
Yes! Check back later Sunday for an update.

GuyFleeqman, 2021-10-03:
Thanks Mike!  Just FYI, I fixed the two missing () as described in other issues, (naota2k3), but as best I can tell it does not change anything...still get the initial 'glitchy' sound, which then generally, but not always, seems to resolve.

Steve

ViennaMike, 2021-10-04:
> Thanks Mike! Just FYI, I fixed the two missing () as described in other issues, (naota2k3), but as best I can tell it does not change anything...still get the initial 'glitchy' sound, which then generally, but not always, seems to resolve.
> 
> Steve

Try the version I just uploaded (10_3_2021). I've tested it extensively on a Pi4 and a Pi Zero W using an older version of the Raspberry Pi OS, and done partial testing using the latest version of the Raspberry Pi OS  and this version is now working fine for me. Let me know how it goes. 

I hope to also provide SD image files to make installing it easier, but they are quite large (> 1 GB), so I'm not sure the best, most cost-effective, place to provide them.

ViennaMike, 2021-10-04:
> Thanks for working on this it’s awesome. I’m in the process of making a talking Audrey 2 using chatterpi. Have you been able to make any progress on the playback issue? I’m running a zero w and am also experiencing the problem.

Try the version I just uploaded (10_3_2021). I've tested it extensively on a Pi4 and a Pi Zero W using an older version of the Raspberry Pi OS, and done partial testing using the latest version of the Raspberry Pi OS and this version is now working fine for me. Let me know how it goes.

I hope to also provide SD image files to make installing it easier, but they are quite large (> 1 GB), so I'm not sure the best, most cost-effective, place to provide them.

ViennaMike, 2021-10-04:
I believe these issues have been corrected in the latest version, uploaded on 10_3_2021. Let me know if there are any remaining issues that you're seeing. If not, I'll close this issue soon.

jamesluke32, 2021-10-04:
Just tested the newest version and it seems to be working as intended. I still need to dial it in for my setup but otherwise the audio comes in clear and the jaw is moving in sync. I can’t seem to flip the open and closed position though.

GuyFleeqman, 2021-10-04:
Just tested it as well...no Joy.  Rasbpi zero; I get to the end of the ALSA errors, and then nothing happens:
.
.
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
if you see ALSA error messages above, ignore them
End of PyAudio initialization
^this is where it sits...no sound at all.

Are we supposed to have both configuration settings files in the chatterpi directory?

ViennaMike, 2021-10-04:
Are you using all the new files? And yes, all.l the .py files and the
config.ini file should be in the Chatter Pi directory. The old config file
is only used when running the control panel and created if not present, so
not important.

What modes are you running? Microphone or file? Ambient on or off? Start,
timer, or pir?

ViennaMike, 2021-10-04:
One more thought: Have you opened the alsamixer via the command line and
made sure that your usb sound card is selected for audio output (and if
using a microphone through it, for audio input as well). Check then by just
playing the vocals or ambient sound file using your audio player.

GuyFleeqman, 2021-10-04:
Hi Mike,
My bad..I'm a goof.  Thanks for the hint above.  I assumed that the config would be for audio files, but it was on microphone.  I changed that to FILES, and the audio works fine!  On to completion..
Many thanks!!

ViennaMike, 2021-10-04:
No problem! Glad it's working for you. I would have set it for FILE, but
forgot to before uploading. To fully test ChatterPi, I need to run 12 test
cases (three possible trigger values, ambient on or off, and file or
microphone). Microphone must have been the setting for the last test before
uploading.

alexM57, 2022-04-20:
Hi there,

I am using a raspberry pi model 3B + and I get the following error message:

ALSA lib pcm. c : 8545 : (snd_pcm_recover) underrun occurred

 when running the core_rpi_nogui.py file which you can find here: https://github.com/Lx37/PyOpenProto/blob/master/pyaudio_protocol/core_rpi_nogui.py. I believe it has something to do with portaudio but I am not sure. I installed portaudio by using configure && make as described here: http://www.portaudio.com/docs/v19-doxydocs/compile_linux.html. But still portaudio is not displayed when entering pip list in the terminal window.

I also copied the files libportaudio.a, portaudio.h and pa_linux_alsa.h into the project folder, but have not yet linked them by using gcc main.c libportaudio.a -lrt -lm -lasound -ljack -pthread -o YOUR_BINARY , as I am not sure what is meant by YOUR BINARY. I also do not know if this is the problem of the underrun error anyway.

Any help would be much appreciated!

Thanks!

ViennaMike, 2022-10-11:
Sorry, I really have no idea. It looks like pyaudio_protocol has nothing to
do with PyAudio, except for a similar name. Have you tried posting your
issue on https://github.com/Lx37/PyOpenProto, since that's the software you
are trying to use?