ViennaMike/ChatterPi #6: GUI + Boot On  Startup Not Working

AndrewDanko, 2021-07-18:
Hi there. I am extremely new to my Raspberry Pi B 4gb Model, so perhaps this is user error but I am having two issues:

1) I tried to run the controlPanel.py however it did not work. The last line of my error message said Syntax Error "(" unexpected.

2) Auto boot wont run, I've followed directions to a T

3) I havent had the opportunity to test the mic yet... but.. when I use the commands:
$ cd /home/pi/chatterpi
$ python3 main.py

it will start up and play the default ambient sound that comes with ChatterPi

there are some other tiny errors in your manual that throws a noob like myself off. The one on page 11 discussing the path for auto-boot is listed as "...system/system..." versus  "...systemd/system..."  May also want to suggest for noobs that to drop the chatterpi.service file, they should open it with "sudo pcmanfm" to allow permissions of dropping it in. 

Overall Mike this is an AMAZING PROJECT and I am so happy you have been kind enough to share... I really hope I can get these issues figured out as I am sweating bullets because this convention is 4 weeks away. 

Hope to hear back soon. If you have a discord or other means of contact, please let me know! I can walk you through what I am doing and maybe catch something I am doing wrong on my end. 

Thanks!!!

ViennaMike, 2021-07-23:
Sorry for the delay in responding, I've been busy with other things. I'll
try to look at this over the weekend. And thanks for the other suggestions
as well.

ViennaMike, 2021-07-25:
On item #1, I can't reproduce that error. See the attached screenshot of
what I get when I run controlPanel.py. You can see the command I entered at
the command line and the panel which then came up. No error. Can you give
me a screen shot of your command and the result? Maybe that will give a
clue.

The good news is that controlPanel.py isn't needed. It's a useful tool, but
you can edit the config file directly.

I'll look at your other issues tomorrow.

AndrewDanko, 2021-07-25:
Ah welcome back! Thank you. You could not have come at a more useful time. I just received my Raspberry Pi Zero W in the mail.. I only have a little less than two weeks left to get this thing working and _I'm beginning to worry_.
Is there a certain version of Raspbian software that you install? I am trying to install the software, but Im having issues with my W showing the display..it continues to blink over and over but doesn't work, even though it works on my RBP 4. 
I'm in the middle of the process right now of trying to do a headless version...but... Putty.exe doesnt seem to be able to find my device. I also installed Bonjour Printing as well to help with this. 

Any help that you can offer to help me get this going would be much appreciated. 

Thank you Mike!

AndrewDanko, 2021-07-25:
Okay, update - I am have the pi zero w installed with the latest version of the OS. Im going to go through the tutorial again and see what happens. I will keep you posted on my progress.

In theory... the SD card from my 4 should be able to work on the zero no problem, right? It's all the same operating system and settings?

ViennaMike, 2021-07-25:
I know little of this and I'm not sure. Apparently the 4 allows a bigger
boot partition than the zero. One source I read.aays it will work
fine,.another says the 4 REQUIRES a larger boot partition and hence won't
work. I think it's worth trying, if it doesn't work, it will show up
quickly.

ViennaMike, 2021-07-25:
Have you got the zero up running headless? I usually make the IP assignment
permanent on my router, so that I always know what IP address my
various PI's have.

Right now, I'm troubleshooting using a Pi 4 (because there's additional
functionality I'm working on that won't all run on a Zero W). I'm running
Raspbian 10 (buster). I believe I've been running the same version on the
Pi Zero W.

Concerning autostart, it looks like I had a typo in the manual and that it
also appears that the proper place to put it has changed to
/etc/systemd/system rather than /lib/systemd/system. (I had dropped the "d"
in the manual, my apologies). Otherwise it's the same, and see
https://www.raspberrypi.org/documentation/linux/usage/systemd.md for more
info.

With this, I'll await further feedback and/or questions from you.

AndrewDanko, 2021-07-25:
Okay I have an update for you. When I switched the SD cards out to the one that i had setup on my RBP 4; it actually booted on startup! I couldnt get it to do this on the RBP4. 

So it seems that I am having the exact issues as I had before on the RBP 4. The servo seems unresponsive... a couple things ran through my mind:

1) Since I am using only a mic for input and a speaker for audio output, do I need to have a trigger/LED setup? 
2) How exactly do you run the control panel? Perhaps I am not doing it correctly. In the manual it says "run control panel". I tried a few things, but  this issue may be user error as I was blindly going at it lol. 

3) Currently the only thing I have hooked up is the servo to their respective pin layouts. 5v+ and ground; pin 18 for the data wire.

ViennaMike, 2021-07-25:
Let me get back to you on 1 and 3. On two, to run controlPanel.py it's like
any other python program. You can run it from an editor, like Thonny or
Geany, or just go to the command line, change to the chatterpi directory,
and type python3 controlPanel.py:

cd chatterpi
python3 controlPanel.py

The control panel should open right up.

AndrewDanko, 2021-07-25:
Also.... when it said "install chatter pi files" into the directory...all I did was create a new folder that says "chatterpi"
then drag and dropped the ambient and vocals folders, audio.py, bandpassFilter.py, config.ini, config.py, controlPanel.py, main.py, maxVol.py, and tracks.py

Also, just trying to troubleshoot.... I changed the permissions to be executed from "Nobody" to "Anyone"

Is there something Im missing?

ViennaMike, 2021-07-25:
OK, on item 1, you need to either edit the config.ini file directly or via
the control panel to set the prop_trigger equal to either TIMER or START.
Timer will only activate the prop every so many seconds (set by the DELAY
value, in seconds). Certainly for testing, and maybe in your application,
it's probably best to set it to START. Then it starts right up.

However, it may be this mode that has an error when using the microphone
that someone else found and that I haven't fixed in the uploaded code yet.
Go to Line 37 in control.py and add a set of parentheses "()" at the end of
the line and then save the change.

ViennaMike, 2021-07-25:
I don't think you're missing anything.

AndrewDanko, 2021-07-25:
Okay great. I will try that now... Also it seemed that  I wasnt running the controlPanel.py correctly (I am brand new to pi/programming but thankfully your guide is well written)

When I DID try to run the code; I got this error:
pi@raspberrypi:~/chatterpi $ python3 controlPanel.py
Traceback (most recent call last):
  File "controlPanel.py", line 140, in <module>
    configuration = ConfigManager(window)
  File "controlPanel.py", line 19, in __init__
    shutil.copy2(configpath, oldpath)
  File "/usr/lib/python3.7/shutil.py", line 263, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.7/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'oldconfig.ini'
pi@raspberrypi:~/chatterpi $ 

Trouble shooting again  I replaced the files so the original permissions would be in place  however i still received the same errors as above

ViennaMike, 2021-07-25:
I didn't do anything special when setting up the folders on their
permissions, but my chatterpi folder is set to:
View content: Anyone
Change content: Only owner
Access content: Anyone

It looks like in your case the control panel program is being denied access
to create the file called oldconfig.ini, which it uses to store the
configuration at the time you open the control panel, so that you can
revert back to it. Change content should be set to either Only owner if you
are running as the owner or to Anyone.

AndrewDanko, 2021-07-25:
Okay I made all of the recommended changes:

1) all file permissions and chatterpi folder are set to be executed by "anyone".

2) I added the "()" parenthesies on line 37 in the config.py

3) I changed the prop_trigger = "START"; also delayed the prop to = "1" and for the eyes, and trigger out I set to "OFF"

....Although nothing has changed, it does feel like I am getting closer... I will upload the string of prompts I get when I try to run the main.py

AndrewDanko, 2021-07-25:
**Prompts when I run ChatterPi:**

pi@raspberrypi:~/chatterpi $ python3 main.py
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=6,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred

ViennaMike, 2021-07-25:
You are getting closer! All of the error messages up until the
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
ones are normal and can be ignored. The underrun error occurs because the
Pi isn't keeping up with processing audio while controlling the servos. Do
you have a speaker or earphones set up to hear audio, and does the jaw move
at all? That can help troubleshoot. Also try changing the buffer_size in
config.ini to 4096 (yes, I know the manual says not to change it, but the
right value seems to change for various people / Pi's).

AndrewDanko, 2021-07-25:
If we can get this working, I will buy you a damn cake haha.

I dont have any motion from the servo at all. The ambient noise will play sometimes. I have a self powered speaker that my pi is connected to...but i actually don't need thsi because I bought a nice little voice changer that will take care of all the sound..If i can get the servo to move while speaking, that will be a massive win.

okay I will change the config.ini and keep you updated. By the way... .when I added the "()"  in the config.py... i did it without quotation marks...and indented first with the "TAB" key, was this correct?

AndrewDanko, 2021-07-25:
Good news, after editing the () I was able to type in the command to get the control panel, and I was able to pul lit up. Now I notice that when I run chatterpi...the ambient audio doesn't start immediately...not sure if this is progress, or not? lol

What should the Prop delay be set to? 0 or 1?

ViennaMike, 2021-07-25:
In control.py, here's what lines 36 and 37 should look like with the
parenthesis added:

    else:
        a.play_vocal_track()

Prop delay shouldn't matter in START mode with MICROPHONE. But turn off
ambient, as it doesn't really make sense if you're activating the mic
immediately and continually for testing. Let me know what, if any, errors
you get, and if you hear your voice or very distorted bursts out of the
speaker, and if the servo moves at all.

Off to dinner now, will check back later, as I have to get back to the PC
anyway to do some other stuff later tonight.

AndrewDanko, 2021-07-25:
Okay, i am also going to recharge on some dinner haha. Thank you so much for being patient and guiding me through this.
I am going to reinstall the original files from the Master.zip again incase I messed with anything...when I start chatter pi now...the ambient audio doesnt play anymore even when it is set to "on"

I had added the () in the config.py on line 37, and not the control. 
I'm going to make sure that after I replace the files to the original, that I make the correct changes in the right folders just incase i got something mixed up. 

Quick reminder for myself
* edit the () in CONTROL.py
----
* set the buffer size to 4096 in CONFIG.py
* Also change prop_trigger=  START
* Source = MICROPHONE
* Check Permissions in chatterpi to "ANYONE" if unsuccessful

AndrewDanko, 2021-07-25:
Okay update... I didnt replace the files yet, but I realized my "Source =" was still set to FILES
I ran it aain and now the light on the soundcard seems to blink, and there is a light feedback noise moving in synch with the red soundcard light. This...may be a good sign? Sounds like a really quiet steam train picking up speed.  

Okay I am going to actuall ywalk away this time and eat dinner haha

AndrewDanko, 2021-07-26:
Good Morning,

So after a fresh start I overwrote all of the chatterpi files and made all of the necessary corrections as seen above...unfortunately...no dice :( 

I realized the ambient sound isn't playing because I went from PIR to START which is good. Ive reset it up several times over. I checked my servo to make sure it works. My mic works on other devices... I'm completely at a loss here.

Here are the prompts that I get when I try to run chatterpi after making those necessary changes. Good thign is that it does create the "oldconfig" folder, and the _pychache_ 

--------
pi@raspberrypi:~/chatterpi $ python3 main.py
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... cut ...]
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=6,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
pi@raspberrypi:~/chatterpi $

ViennaMike, 2021-07-26:
OK. Good and bad news is that there's no errors, but it's just sitting
there. Let's check a few things to try to isolate the remaining issues.
1) Do you still have the light blinking on the USB sound card, indicating
it's getting input?
2) Send me a listing of your current config.ini file, so I can check it and
be a second set of eyes
3) Let's make sure that you are getting sound in from the mic, without
using chatterpi, to be sure that's not the problem. Record a 10 second
sample using this on the command line:
arecord -d 10 -c 2 test.wav

You should see something like this and then get the prompt back after about
10 seconds. Also, your soundcard light should blink during the recording:
pi@Pi4:~ $ arecord -d 10 -c 2 test.wav
Recording WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Stereo

Then enter:
aplay test.wav

The sound should play back and you'll see something like this in the
terminal:
pi@Pi4:~ $ aplay test.wav
Playing WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Stereo
pi@Pi4:~ $

Let me know how this goes.

AndrewDanko, 2021-07-26:
Okay, you are a lifesaver. I have been searching all over google because I suspected the issue was coming from the mic...I saw some had similar issues on a forum from years ago such as this one:
https://www.raspberrypi.org/forums/viewtopic.php?p=314611#p314611
Tried looking for directions to test the mic on google, but none have been as simple and easy to explain as yours....so THANK YOU for being so kind and patient. Even though I am trouble shooting this and overcoming a learning curve; it has been enjoyable. You may have opened up a new hobby for me!

**I tested the mic** and it definitely seems to be an issue with the input. What I found:

*The light stays lit, but does not blink. 

*When I hit record and played it back, it didn't capture my voice. It sounded like it picked up on background ambience of my fan blowing? Or it could have just been that feedback sound when your speaker is trying to play something. Either way, no voice was captured.

*When I right click the speaker icon in the top right... it does list "USB Audio Device"  for input, but it only shows it as listed. I later realized I should be seeing an adjustable slider for mic sensitivity, right? 

This is the soundcard that I have purchased:
https://www.adafruit.com/product/1475
I noticed they said they have only tested it with "Raspbian Wheezy & Jessi." It mentioned a tutorial....but I couldn't find one online, and it didn't come with anything. 
---
Incase you still wanted to look at the config here it is!
[SERVO]
servo_min = 1050
servo_max = 1250
min_angle = 90
max_angle = 0

[CONTROLLER]
style = 1
threshold = 2500
level1 = 1500
level2 = 2500
level3 = 3500
filtered_level1 = 1000
filtered_level2 = 2500
filtered_level3 = 4000

[AUDIO]
buffer_size = 4096
source = MICROPHONE
output_channels = BOTH
mic_time = 30
ambient = ON

[PROP]
prop_trigger = START
delay = 10
eyes = ON
trigger_out = ON

[PINS]
jaw_pin = 18
pir_pin = 23
eyes_pin = 25
trigger_out_pin = 16

AndrewDanko, 2021-07-26:
Quick Update; I followed this tutorial, and turned turned the mic gain up to 100 using the alsamixer. The light blinks now as soon as it starts recording, but still no audio playback of what I am trying to record

ViennaMike, 2021-07-26:
Okay, so first we need to make sure your mic and stuff is working before
worrying about ChatterPi. The good news is that I'm using that same sound
device, so I know it can work.

Are you using the ALSA mixer that you start from the command line to set
the volume, as opposed to the little speaker in the GUI? DO use the ALSA
mixer for volume setting, as I get strange results sometimes when adjusting
the volume on the GUI. Also make sure that the right device is chosen for
the speaker output. In the mixer, hit F6 to select sound card and  select
USB Audio Device if you haven't, and check that both the  speaker and mic
levels are set to at least the top of the green. You can (and should) use
the GUI speaker icon (right click on it) to check that both Audio Outputs
and Audio inputs are set to USB Audio Device. If you don't have the GUI up,
let me know and there are command line commands that can check this and set
this, but I'd have to look them up.

AndrewDanko, 2021-07-26:
Hm yeah, I have tried both the gui and the alsamixer command. With alsamixer I did set my mic up to about mid way, and adjusted the capture to midway. Original setting was pretty low.  I was hoping that would fix it, but unfortunately It didn't seem to do anything. The output of the sound card seems to be working fine.. input seems to be not working...could this be a bad sound card?

ViennaMike, 2021-07-26:
It certainly could be a bad sound card, but I wouldn't leap to that. I'm
using a cheap phone set of earbuds with microphone for testing. If you have
anything like that around, maybe test with a 2nd mic.

Is the test.wav file being created?

If you are getting a test file to play and VLC Media Play is installed, you
can also see if this plays back the recorded file or install Audacity (go
to preferences under the raspberries in the upper left of the gui window,
select Add / Remove software, and search for audacity) and see what it
thinks the file look like and will play. Audacity will also record, so
another way to test.

Finally, can you tell me what you get when you type lsusb -t in the command
line? You should get something like
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 3, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 4: Dev 3, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 4: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 4: Dev 3, If 3, Class=Human Interface Device,
Driver=usbhid, 12M
but your numbers may differ. Also type arecord -l and let me know what you
get. It should be something like:
**** List of CAPTURE Hardware Devices ****
card 2: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

AndrewDanko, 2021-07-26:
I tried two mics so far. I'll try another and see what happens. Im installing audacity now. Good news is that Im familiar with audacity. Didnt know I can get it on pi, so that is pretty nice.

pi@raspberrypi:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 6, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 3: Dev 6, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 3: Dev 6, If 2, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 3: Dev 6, If 3, Class=Human Interface Device, Driver=usbhid, 12M
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $

AndrewDanko, 2021-07-26:
Installed Audacity and tried it out on all 3 mics, same outcome: just mild white noise

ViennaMike, 2021-07-26:
I'm out of ideas, I'm afraid. If you can figure out your mic input problems
I'm happy to help with any ChatterPi issues, but I don't know why you're
having sound input issues.

AndrewDanko, 2021-07-26:
Well.... I did just stumble onto this: 
https://www.raspberrypi.org/blog/new-raspberry-pi-os-release-december-2020/

could Pulsa be the issue? The comments seem to have some useful info...but I didnt quite udnerstand it.

Can you tell me the exact Raspberry Pi OS that you use for your Pi Zero W when you use Chatter Pi?

I'm going ot try and install whatever you have and see if that will fix it.  Thank you for all of your help :)

ViennaMike, 2021-07-26:
I had to dig out that Pi and check. It's running version 10 (buster),
version 10.4 to be precise, with linux kernel 4.19. So from your note and
this table, it seems to predate the move to Pulse. I've no idea if that
could be a problem or not. I'm running 10.9 on my Pi 4, which should have
Pulse, and the mic works on it.

AndrewDanko, 2021-07-27:
GUESS WHAT??!!!

I did a clean install of Buster, andddddd.... nothing... So I was about to call it quits..and I tried yet another mic.. and THEN... IT WORKED... 

In my search I noticed that PulseAudio seems to be a headache...and even more stranger...every single issue I have seen on forums, comment sections, etc about the mic not working as a record function...is always from this year or the tail end of December 2020 when PulseAudio came to light. I'd hang onto that versio nif you have it running...seems like a lot of people were getting frustrated with this.

I am now working on getting ChatterPi all set back up.... the first test I just did...didn't seem to trigger the servo. My config.ini is the same that I sent you before. Im going to keep messing with it and see if I can get it going.

I'll keep you posted on what I find out...hopefully the perspective of a very determined noobie will  help future people wanting to to set up your project.

AndrewDanko, 2021-07-27:
Just wanted to mention one other thing as it may be significant. The servo Im using is this one...
https://www.adafruit.com/product/2307
If you read the bold areas...could that make a difference? It said somethign about 5v logic?... I have everything pinned to the 5V side?

ViennaMike, 2021-07-27:
That could be a problem. The Pi doesn't put out 5V on the GPIO pins, it
only puts out 3.3V, and the description seems to be pretty clear that this
servo needs 5 volt logic on the signal pin as well as 5V power. There's
more info and products that will shift here:
https://learn.adafruit.com/neopixel-levelshifter

Back on the ChatterPi front, do you hear anything out of the speaker when
trying it, or see any messages such as underrun or others after the
usual string of alsa error messages when starting?

One thing you can try is change the source to FILES, prop_trigger to TIMER,
and delay = 5 and see if the supplied Wave file 1) plays out the speaker
using ChatterPi and 2) if the servo moves. That would indicate the output
side is working or partly working. That says play the wave file over and
over with a 5 second delay in between. If you hear very garbled sound, try
both 4096 and 1024 for the buffer_size and let me know if there's any
change.

Your info on the change to Pulseaudio is quite interesting, and may explain
some different issues that others have had and I can reproduce with garbled
sound because the audio processing isn't keeping up. It may be that adding
Pulseaudio to the mix is slowing things down. But that problem is mostly
fixed with the larger buffer size. Something I'll have to look into more to
see if Pulse is the issue there and if I can disable it for this
application if it is.

ViennaMike, 2021-09-18:
Closing as ChatterPi software issues addressed with the exception of the buffer underrun issue, which is covered as issue #4.

closed 2021-09-18 by ViennaMike