ViennaMike/ChatterPi #8: PIR issue and cannot run on startup

danog14, 2021-10-04:
Hello, in the new version, the PIR only seems to start the first event and then it seems to go into timer mode and I have it set to FILES and PIR.  Also it will not work on startup after putting the chatterpi.service in the etc.systemd.system folder.  I did it with permission as well.

ViennaMike, 2021-10-04:
I can't duplicate your first case. When I test (just retested several
times), the PIR mode with FILES works as it should. The audio file plays
when the PIR is triggered, and then doesn't play again until the PIR is
again triggered AFTER the delay setting (in seconds) occurs. I once thought
I was having this problem, but in that case it turns out the PIR was being
triggered fairly often without my knowing it, so almost as soon as the
delay timer expired the sound would start again. Make sure nothing else is
accidentally triggering the PIR and try different value for delay and let
me know what happens.

I'll look into the autostart issue next.

ViennaMike, 2021-10-04:
For startup, are you using the old, original version of chatterpi.service?
I ask because I accidentally forgot to include it in the updated repository
(my mistake). If so, try changing:

WorkingDirectory=/home/pi/chatterpi
to
WorkingDirectory=/home/pi/ChatterPi

and enable it again. Let me know how that works. If that's the problem,
I'll update the repository with the fixed version. Thanks!

danog14, 2021-10-04:
I got the PIR to work once or twice but then it just moves onto the next track most of the time after the allotted time out for the PIR.  Nothing is tripping it as I have it completley covered.  I just did a fresh install of OS and chatterpi.  I don't want to check the startup until I have this figured out though.  I have tried 3 different PIR sensors.  Not sure what the issue is.  It seemed to work with the older version, but just not with the newer version.  I tried changing the PIR time as well, but it is still doing it

ViennaMike, 2021-10-04:
do you have multiple tracks or just one, and is ambient mode on or off? I
just want to set up the same conditions when I test.

danog14, 2021-10-04:
I have 11 tracks and ambient is on currently.  It just seems erratic.  Sometimes the track plays before the ambient sound even comes on occasionally the PIR will work.  It usually works for the very first time and after the first trigger it does not loop back and wait for another trigger.  I got it to work one time, but then the very next track again just started playing, then the next etc.

danog14, 2021-10-04:
almost always the first trigger will wait for the PIR with the ambient music playing.  Once the first event is triggered though the PIR doesn't seem to work.

ViennaMike, 2021-10-05:
Hold off on the run on boot, there does seem to be an issue there, and I
haven't found it yet.

Regarding the other issue, I still can't duplicate it. I created 10 vocal
files, turned ambient on, with PIR, and I successfully cycled through all
10 tracks over 2 times, with the ambient track playing in between each
time, and the audio only starting when the PIR was triggered. So I'm not
sure what's going on there. I will mention that the code is only written to
support up to 10 vocal files (must be named v01.wav through v10.wav), not
11. Although that doesn't seem to fully explain your issues.

danog14, 2021-10-05:
I wi maybe buy a different PIR and see if it is an issue.  I have been using them in various scenarios for my project do maybe something went bad with them.   Thanks.  Let me know about the run on boot as well when it is fixed.  Thanks so much and awesome project.   Mine is a talking billy (jigsaw) from saw.  I look have a pololu controlling the eyes and head/neck but I am running that in a different system and trigger as I could not get it to trigger correct from the same PIR as the chatterpi is running, but it works just as well with 2 PIRs.   Thanks again

ViennaMike, 2021-10-07:
> I think i got the PIRs working again. I adjusted some of the sensitivity
> and timing. Not sure if that worked or just restarting a couple of times,
> but now if I leave the room and come back in it seems to work. Now I will
> just wait for the start on boot to be fixed to implement that. Thanks so
> much for this program, it is great.

Glad you got the PIR working! I've been pulling my hair out and spending
too many hours on the autostart issue and I finally think I know what may
be happening, but it's late in the evening, so I will have to get back to
it later this week. The Raspberry OS switched awhile back to using Pulse
Audio at some point, which PyAudio uses, BUT it doesn't start up until the
graphical interface startup process, so normal Chron jobs and Systemd
approaches don't work, because they try to launch too early. I'd have
thought a delay would fix it, but it doesn't seem to. But one can specify
scripts to run in the graphical interface startup process. There's one or
two discussion threads where this fixed their Pulse Audio issues relating
to startup scripts. Hopefully this will fix the problem. Stay tuned.

danog14, 2021-10-11:
Okay, I have found another issue.  when I put my prop together and started the PI, it did not work.  I plugged the monitor and keyboard and mouse in again and it worked.  This happens over and over and I guess they have an issue with the HDMI and booting headless without the HDMI plugged into a monitor.  I cannot get it to work without the HDMI plugged into the monitor now.  I have tried a few solutions trying to fix it, but have not found one that works yet.  Any thoughts on how to fix this?  It comes out of the soundcard or 3.5 jack just find when the monitor is on.

ViennaMike, 2021-10-13:
PIR issue was a user hardware problem. Startup issue fixed with the 10_11_21 update, so closing the issue.

closed 2021-10-13 by ViennaMike