ViennaMike/ChatterPi #10: Wont start at boot?

GuyFleeqman, 2021-10-17:
Hi Mike,
Have the latest code I believe.  I have done all in the para describing  run automatically on boot, but the program doesn't start.  Following reboot, I open a terminal and see this: 

Running at boot
Traceback (most recent call last):
  File "/home/pi/chatterpi/main.py", line 10, in <module>
    c.update()
  File "/home/pi/chatterpi/config.py", line 40, in update
    SERVO_MIN = int(cfg['SERVO']['SERVO_MIN'])
  File "/usr/lib/python3.7/configparser.py", line 958, in __getitem__
    raise KeyError(key)
KeyError: 'SERVO'
pi@raspberrypi:

I've also tried a variety of methods to get this to run, cron, localrc, etc...with no success.  Appreciate your thoughts.

ViennaMike, 2021-10-18:
I'm away from my test setup for awhile, so I won't be able to look at this
this week.

ViennaMike, 2021-11-23:
Are you using the method I describe in the manual (using autostart)? I just retested it and I don't get the error.  

Apparently an issue can arise with cron (and perhaps other approaches) runs main.py in a different working directory, hence the link to the config.ini file doesn't work. There are other problems with some of the other approaches as well for programs like ChatterPi that expect the desktop to be up and running.

ViennaMike, 2022-03-24:
Haven't had any follow-up on for four months, so closing. If still problems, please open a new issue.

closed 2022-03-24 by ViennaMike