ViennaMike/ChatterPi #11: Audio playback stops intermittently

GuyFleeqman, 2021-10-18:
While running the program; three vocal tracks, one ambient track, playback seems to work until it stops intermittently with the following trace:
.
.---standard ALSA error messages here....
if you see ALSA error messages above, ignore them
End of PyAudio initialization          (audio stops with following error trace):
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pigpio.py", line 245, in _set_state
    if value != self.factory.connection.get_PWM_dutycycle(self.number):
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1511, in get_PWM_dutycycle
    return _u2i(_pigpio_command(self.sl, _PI_CMD_GDC, user_gpio, 0))
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1011, in _u2i
    raise error(error_text(v))
pigpio.error: 'GPIO is not in use for PWM'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/chatterpi/audio.py", line 97, in filesCallback
    self.jaw.angle = jawTarget
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 447, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 132, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1804, in angle
    self._min_value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 447, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 132, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1642, in value
    ((value - self._min_value) / self._value_range)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 285, in <lambda>
    lambda self, value: self._set_state(value),
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pigpio.py", line 248, in _set_state
    raise PinInvalidState('invalid state "%s" for pin %r' % (value, self))
gpiozero.exc.PinInvalidState: invalid state "525" for pin GPIO18

Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/home/pi/chatterpi/audio.py", line 120, in cleanup
    normalEnd()
  File "/home/pi/chatterpi/audio.py", line 117, in normalEnd
    self.jaw.angle = None  
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 447, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 132, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1798, in angle
    self.value = None
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 447, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 132, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1637, in value
    self.pwm_device.pin.frequency = None
AttributeError: 'NoneType' object has no attribute 'frequency'
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/home/pi/chatterpi/audio.py", line 120, in cleanup
    normalEnd()
  File "/home/pi/chatterpi/audio.py", line 117, in normalEnd
    self.jaw.angle = None  
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 447, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 132, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1798, in angle
    self.value = None
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 447, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 132, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1637, in value
    self.pwm_device.pin.frequency = None
AttributeError: 'NoneType' object has no attribute 'frequency'
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/home/pi/chatterpi/audio.py", line 120, in cleanup
    normalEnd()
  File "/home/pi/chatterpi/audio.py", line 117, in normalEnd
    self.jaw.angle = None  
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 447, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 132, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1798, in angle
    self.value = None
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 447, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 132, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1637, in value
    self.pwm_device.pin.frequency = None
AttributeError: 'NoneType' object has no attribute 'frequency'
pi@raspberrypi:~/chatterpi $ 

This error trace seems pretty consistent, ie, when it quits, the trace always starts at the same point.

AlphaEcho11, 2021-10-18:
What triggers do you have set currently? Timer, PIR sensor, etc?

I initially had some similar issues running in Timer trigger, but that cleared completely during PIR sensor trigger. 

Each of those errors are returning to GPIO libraries, pigpio especially at the onset. Remove and re-add?

ViennaMike, 2021-10-24:
I believe you are correct, the presence or absence of an actual servo being attached shouldn't matter. 
Unfortunately, I don't know why you are getting pigpio errors. I might suggest removing and reinstalling python3-gpiozero, then pigpio:
$ sudo apt install python3-gpiozero
Note: you may find this and some of the other modules below are already installed on newer versions of
the Raspberry OS. It won’t hurt to try to install them again. Do NOT remove files that the install process
suggests can be safely removed. Apparently, THEY CAN’T!
To install PiGPIO, type:
$ sudo apt install pigpio
$ sudo systemctl enable pigpiod
$ sudo apt install python3-pigpio

ViennaMike, 2022-03-24:
Seems to be an issue with gpiozero or pigpio, rather than with ChatterPi itself, and have not see a follow-up in 5 months, so closing.

closed 2022-03-24 by ViennaMike