ViennaMike/ChatterPi #14: Buffer underrun

diveblue, 2023-09-08:
Using 5 files of 6-10MB each, PIR trigger. function well for about 2 hours, then I get several 
buffer underruns reported and playback / trigger sensing stops

ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred

ViennaMike, 2023-09-09:
What type of pi are you using?

arwpc, 2023-09-10:
Pi Zero W or a Pi 4. Thanks! I love the software!

diveblue, 2023-09-11:
Pi zero w.

Incidentally, the image did not work so I had to do the manual install of
the prerequisites

arwpc, 2023-09-11:
Ditto that. I’m trying to keep the cost down from the Pi4 because I’d like
to network several together. ;)

ViennaMike, 2023-09-21:
OK, this one will be a tough one. There's no doubt that ChatterPi really pushes a Pi to its limits. The issue is clearly related to taking too long in the audiopy callback function that determines the jaw movements, but I have not found a way to make that run any faster. Any suggestions for further optimizing that code welcomed. 

Try increasing the BUFFER_SIZE parameter to 16384. In the past, this seemed, in my setup, to elliminate the problem on the Pi4.
Changing the CPU governor to performance mode may also help, but it wasn't sufficient by itself. The downside of a larger buffer size is that it introduces delay in the output. 

Hope this helps!  I started a separate issue regarding the image file not working.

closed 2023-12-04 by ViennaMike