ViennaMike/ChatterPi #16: Variables "now" and "data" not defined in mic callback which throws an error during runtime

codeobsession, 2024-02-09:
Tried running the code with the test setup for the chatterpi skull project. Servo motion is as expected from file input. When I switch to mic input, the interpreter complained about a couple errors in the code:

1. "now" is not defined in the mic callback
2. "data" is not defined in the mic callback

It looks like code from the files callback may have been cut and pasted into thee mic callback and not tested?

Also, the conditional in the mic callback testing for two channels should be removed because the number of channels is set to 1 right above it so it will never get called.

ViennaMike, 2024-02-13:
Thanks for the bug report. You are correct that I didn't have a microphone set up handy, so didn't fully test the last update. I've fixed and tested it todayt, and will upload the fix to GitHub in a day or two.

ViennaMike, 2024-02-13:
The code has been corrected, tested, and the source code uploaded. Please let me know if you get a chance to test it. I'll keep this issue open until I get a chance to also update the image file.

codeobsession, 2024-02-14:
I tested the new audio.py file and the mic callback worked as expected.

ViennaMike, 2024-02-15:
Thanks! At some point all was tested, but as I mentioned, when I last updated audio.py I didn't test the mic update. Glad all is good now. If you find any other bugs, be sure to post them. 

I've also now uploaded the fixed img. file, so I'll close this issue.

closed 2024-02-15 by ViennaMike