ViennaMike/ChatterPi #17: No Manual Found on Github

maddog0487, 2024-02-25:
Hello Mike,

Looking for the manual for ChatterPi, figured out quite a bit however documentation would be of great help.

I have it working for the most part however, I need your instructions on startup.

Thanks!

ViennaMike, 2024-02-26:
The [user's manual](https://github.com/ViennaMike/ChatterPi/blob/master/Chatter%20Pi%20Users%20Manual%20v09%2010_3_21.pdf) is on GitHub under the project, one level up from the src folder. 

p.s., I'd love to hear/see more of how anyone is using ChatterPi. Post comments on my blog: https://www.mcgurrin.info/robots/690/

maddog0487, 2024-02-26:
Sorry Mike, NOOB alert.  I did experience an error using the "START" trigger, after a few minutes it crashed, it didn't like line 127 of the audio.py program something about the PWM frequency and Nonetype?  I just used the TIMER instead and set the mic input for a really long time and shorted the trigger input.  I wil post something shortly on your blog page.

maddog0487, 2024-02-26:
Resolved!

closed 2024-02-26 by maddog0487