ViennaMike/ChatterPi #18: My skull is open at the end

Zafp, 2024-05-24:
Currently I have this problem that at the end of the audio the skull leaves its mouth wide open, during the audio it talks normally, is there any way to make it stay closed at the end?

current servo config is:
servo_min = 800
servo_max = 1100
min_angle = 0
max_angle = 90

Ive been trying to change these to no progress.

ViennaMike, 2024-05-24:
The mins and max's won't help with this, they just set the limits. The jaw will stop at whatever was the last audio level. Can you add a brief bit of silence to the end of your audio track? 1/2 - 1 second should do it. If so, let me know how that works.

Zafp, 2024-05-24:
Sure, will try it

Zafp, 2024-05-24:
It was but the opposite, seems that I messed up and open is closed and closed is open, so if it ends on a silence it opens and if it ends on sound it keeps shut

ViennaMike, 2024-05-24:
Great! That's easy then. This is actually covered in the manual in the detailed set-up questions. Different skulls are set up differently. If you are getting the opposite of what you should get, change the min and max angles to be:
min_angle = 90
max_angle = 0

Let me know if that works, and if so, I'll close the issue. 

And if you don't mind, I'd love to see photos or videos of your project. You can post them as part of a comment on my blog: ([https://www.mcgurrin.info/robots/690/])

Zafp, 2024-05-25:
I think I tried that one, but will try again on Monday since its on the friends workshop and not gonna be there till Monday.

Zafp, 2024-05-27:
Odly, it acts the same wether I invert the values or not

ViennaMike, 2024-05-29:
OK, I don't have my skull hooked up presently, so let me put together a little test rig and install ChatterPi on another Pi and look into it. It will probably be a few days.

Zafp, 2024-05-29:
No pressure. Thanks for going the extra mile, sincerely

ViennaMike, 2024-05-31:
OK, I did indeed have a bug in the code. If you replace the audio.py file with the new one on GitHub, that should fix the problem. Now reversing the min and max angles in the ini file should indeed reverse the direction on the servo, as it should. I also added a new audio file to the vocals folder, v02.wav. This file consists of a loud tone, followed by silence, then the tone again, then silence. That should be convenient for testing out your setup. 

When you get a chance, let me know if this works, and I can close the issue.

Zafp, 2024-05-31:
It works perfectly, thansk a lot, I will upload a video when I have a chance.

ViennaMike, 2024-06-03:
Bug fixes tested and uploaded, along with a new image file with the fix. Closing issue.

closed 2024-06-03 by ViennaMike